OS-EDA/Course #2: Which `pandoc` version is needed to compile/build slides?

sergeiandreyev, 2025-01-20:
When trying to compile slides on Ubuntu 22.04 I have the following error messages and the PDF files are not getting generated:
```
$ ./generate_slides.sh
Unknown extension: rebase_relative_paths
...
```
The Pandoc version is:
```
$ pandoc --version
pandoc 2.9.2.1
Compiled with pandoc-types 1.20, texmath 0.12.0.2, skylighting 0.8.5
```

sergeiandreyev, 2025-01-20:
also, if I remove the `--from=markdown+rebase_relative_paths` option from the command line the next error is showing up:
```
Unknown input format markdown_strict
```

ThorKn, 2025-01-26:
It is tested with 3.4 and should run with >= 3.4

sergeiandreyev, 2025-01-27:
ok, then the compilation will not run by default even on the IHP external Open source server w/ Ubuntu 24.04.1:
```
$ pandoc --version
pandoc 3.1.12.2
...
```

ThorKn, 2025-01-27:
That might be true, but it is no requirement to run on a vanilla Ubuntu. And pandoc >=3.4 can easily be installed on any Ubuntu.
I would like to close this issue.

sergeiandreyev, 2025-01-27:
ok, good

closed 2025-01-27 by ThorKn