OS-EDA/Course #22: 2.5D Viewer Script

rslawson, 2025-02-12:
I'm not quite sure where you want this in the repository, but here's the 2.5D Viewer script that I used in the class:
```drc
z(input(1, 0), 0.1 .. 0.2, like: "1/0", name: "Active")
z(input(5, 0), 0.3 .. 0.4, like: "5/0", name: "Poly")
z(input(6, 0), 0.2 .. 0.5, like: "6/0", name: "Contacts")
zz(like: "8/0", name: "Metal1") do
  z(input(8, 0), 0.5 .. 0.6)
  z(input(8, 2), 0.5 .. 0.6)
end
zz(like: "10/0", name: "Metal2") do
  z(input(10, 0), 0.7 .. 0.8)
  z(input(10, 2), 0.7 .. 0.8)
end
z(input(19, 0), 0.6 .. 0.7, like: "19/0", name: "Via1")
z(input(29, 0), 0.8 .. 0.9, like: "29/0", name: "Via2")
zz(like: "30/0", name: "Metal3") do
  z(input(30, 0), 0.9 .. 1)
  z(input(30, 2), 0.9 .. 1)
end
z(input(49, 0), 1 .. 1.1, like: "49/0", name: "Via3")
z(input(50, 0), 1.1 .. 1.2, like: "50/0", name: "Metal4")
z(input(66, 0), 1.2 .. 1.3, like: "66/0", name: "Via4")
z(input(67, 0), 1.3 .. 1.4, like: "67/0", name: "Metal5")
z(input(125, 0), 1.4 .. 1.5, like: "125/0", name: "TopVia1")
z(input(126, 0), 1.5 .. 1.6, like: "126/0", name: "TopMetal1")
```
This cannot be imported and must have some extra XML metadata at the top in order to be used by the viewer, so it should be copy+pasted into the editor.

ThorKn, 2025-02-18:
Would you mind giving this a separate repository? Maybe in your Github?  
Then the EDA course slides could link to that repo.

ThorKn, 2025-02-18:
Forgot to tag you. Here we go: @rslawson

rslawson, 2025-02-18:
Sure thing, I'll do that in just a few minutes.

rslawson, 2025-02-20:
Ah my bad, I made a typo there. Meant to spell that as "I'll get distracted immediately and then forget to do it." Here you go: https://gist.github.com/rslawson/fecc97d8731da95204ed59b00571582f

ThorKn, 2025-02-20:
Because of your script, there is a new advanced training in chapter 05. Thanks a lot.
[https://github.com/OS-EDA/Course/blob/main/Chapter_05_PDK/c05_training_advanced.md](https://github.com/OS-EDA/Course/blob/main/Chapter_05_PDK/c05_training_advanced.md)

closed 2025-02-20 by ThorKn