OS-EDA/Course #33: Chapter 8 rework

ChristianWittke, 2025-02-14:
Rework Chapter 8 completely

ThorKn, 2025-02-18:
Additional info: A lot of information in Chapter 08 went into earlier chapters and trainings.
So, Chapter 08 should be reviewed about this. Maybe some slides can be fully deleted.

ThorKn, 2025-02-22:
After reviewing Chapter 08 again (two times), I don't want to change anything in this chapter. All the information is exactly where it should be.

The reason this appeared as mostly doubles in the testrun of the course are
- intense and forwarding questions of the participants with a lot of pre-knowledge, asked in earlier chapters.
- presentations by Philipp and Rene about topics from chapter 08.

If neither of these wouldn't have happened or someone does the course at home, the slides are still necessary. Additionally, they can be skipped if the information is given ahead of chapter 08.

I don't reduce Chapter 08 therefore and close the issue.

closed 2025-02-22 by ThorKn